betaflight/betaflight-tx-lua-scripts #518: Bug report - Radiomaster GX12 doesnt run BF lua v1.7.0

shyam4learn, 2025-02-06:
1. Describe your problem;
When trying to load Betaflight setup lua script, compilation fails with "syntax error: scripts.lua","bt") error File not found. After that, the radio is basically unusable until I restart it.

2. Include ways to reproduce the problem;
Load BF lua contents onto the GX12 internal memory as it does not have SD card. Go to Tools and run "Betaflight Setup" script.

3. Provide as much information as possible about your hardware and software, including:
Radiomaster GX12  running EdgeTX v2.11.0-selfbuild Git# d5552c80
ExpressLRS v3.x

[image]

[image]

pfeerick, 2025-02-07:
I would suggest first update the firmware on the transmitter to a more recent [nightly EdgeTX build](https://buddy.edgetx.org/#/flash?version=nightly&filters=includePrereleases&target=gx12) for the GX12, and also using the [nightly BetaFlight lua scripts package](https://github.com/betaflight/betaflight-tx-lua-scripts-nightlies/releases). The firmware update should fix your issue, but the updating the lua scripts also won't hurt as there have been several fixes and updates since the 1.7.0 package was released.

shyam4learn, 2025-02-07:
@pfeerick  i did try the latest nightly build ([v20240517.17](https://github.com/betaflight/betaflight-tx-lua-scripts-nightlies/releases/tag/v20240517.17)) for Lua scripts and I get the same error still and same unusable radio behavior on exiting the script. 
Also, as far as nightly EdgeTX build is concerned, my radio is already on a nightly build as the latest public release is 2.10.6, my radio is 2.11.

klutvott123, 2025-02-07:
are you downloading the zip file named "source code"? If so, scripts.lua won't exist. It has to be the "betaflight-lua-scripts-tx_whatever.zip"

shyam4learn, 2025-02-07:
@klutvott123 oh yeah, I am not downloading the source code, I am downloading the actual zip file and it does have the actual lua.

klutvott123, 2025-02-07:
@shyam4learn we have have had similar reports before, and in those cases replacing the SD card have fixed it, but seems like that's not an option in your case

shyam4learn, 2025-02-07:
Yup exactly... I did see some older reported bugs and in most cases changing the SD card fixed the issue

pfeerick, 2025-02-08:
> Also, as far as nightly EdgeTX build is concerned, my radio is already on
a nightly build as the latest public release is 2.10.6, my radio is 2.11.

Yes, you are on a build for 2.11, but you are not on a recent nightly -
your build is clearly from October 2024 ;) There have been significant
changes to the firmware since then, with some specific fixes for the GX12.

shyam4learn, 2025-02-08:
@pfeerick I reached out to radiomaster and they gave me the instructions to update the firmware. I will give that a try and update you guys. If all works, I will close this issue.

shyam4learn, 2025-02-09:
@pfeerick  after updating EdgeTX version to the latest nightly on my GX12, it fixed the issue. Sorry for the trouble and thanks for your help. I am closing the issue.,

closed 2025-02-09 by shyam4learn

gunn-cyber, 2025-10-25:
Hey I’m having this issue as well. Can you explain this process in better detail. Like how Radiomaster instructed you to download new firmware. I’m trying this on edgetx v 2.11.3 with bf lua 1.7.0. I get the same error. My sound packs don’t work properly after updating. I no longer get the welcome message, when powering up my radio. I’ve done this on a frsky radio in the past. Had zero issues then. I guess I’m missing something with the nightlie build. I’m new to elrs and new to figuring things out on this radio. Seems a bit more complicated than my previous radio. If anyone could walk me through this process in great detail, I’d truly appreciate it, thanks for the time. Maybe some links to the firmware, nightlie, etc. sorry I’m not familiar with this stuff yet.

shyam4learn, 2025-10-25:
These were the instructions
_In the off state or no power state, plug in the USB, hold down the SG button on the left shoulder, and then press the power button again, it will automatically enter the DFU mode, you can directly use EdgeTX Buddy to update the firmware, you can easily update the firmware in this way.

When you use a Windows PC to update the firmware, you need to replace the driver with the DFU device STM32 Bootloader, otherwise EdgeTX Buddy will not recognize the STM32 chip.
Check out this description and download the ImpulseRC Driver Fixer driver installer:
https://edgetx.org/faq

You can then smoothly update the Nightly firmware of GX12 using EdgeTX Buddy_.

But honestly 2.11.3 should have worked. Did you update the SD card contents?

gunn-cyber, 2025-10-25:
I did update the sd card contents. I think I have the correct drivers. I will try this again in a few. What I’m confused about is the nightly contents. Where do I find this? Do you happen to have a recent link to this? Thanks

gunn-cyber, 2025-10-25:
I reflashed, and sd card contents again. Do I need to remove bf lua before all of this, and replace it with fresh install? I’m still getting the same errors. The only change I’ve noticed, is my welcome voice is back. I’m also not seeing the nightly contents, maybe this is my problem? Sorry I don’t really understand all of this. This is pretty new to me.

gunn-cyber, 2025-10-25:
Found it, let’s see if this works.

gunn-cyber, 2025-10-25:
Got it working finally! Thanks!