betaflight/betaflight-tx-lua-scripts #520: Issues Navigating through the Menu on the Jumper T15

PhreddStar, 2025-02-14:
My Jumper T15 (vr: 2.10.1)is experiencing some issues when running the latest Betaflight Lua script. The first issue I noticed was the screen resolution problem which has already been addressed in a previous "issue report". Secondly, I am unable to navigate through the menus after launching the BF CMS script. The Pitch and Roll stick seems to have some affect when trying to move from menu to menu. I'm not sure if this issue has been reported..... If it has can someone direct me to the solution page? Thanks in advance!!

klutvott123, 2025-02-15:
Which rx/tx are you using?

PhreddStar, 2025-02-16:
I am currently using the Jumper T15(vr.2.10.6) and a ExpressLRS RX24T 2.4G receiver.

pfeerick, 2025-02-16:
Um, I thought the CMS script was for Crossfire/Tracer only? Hence the...

**!! IMPORTANT: TBS Crossfire/Tracer only !!**

... In the repo readme?

PhreddStar, 2025-02-16:
Understood! Pardon me. I am new to building drones and BetaFlight but I aim to learn. Maybe some "if" statements can be added to prevent new users from experiencing this in the future. Thanks

klutvott123, 2025-02-16:
It should work for elrs too now https://github.com/betaflight/betaflight/pull/12308
What's the framerate?

PhreddStar, 2025-02-17:
What specific framerate are you referring to?

klutvott123, 2025-02-17:
the framerate of your radio link

PhreddStar, 2025-02-17:
I actually have the Baudrate set to 5.25M for the internal RF crossfire rate.

klutvott123, 2025-02-17:
I'm talking about the packet rate that you can find in the elrs lua script.

PhreddStar, 2025-02-17:
250Hz(-108dBm)

klutvott123, 2025-02-18:
try 500Hz

klutvott123, 2025-02-18:
From the PR linked to above "Note that this only works at 333Hz and above or D250 and above."

PhreddStar, 2025-02-18:
Thanks for the suggestions and I must say changing to 500Hz made a drastic improvement of the response of the sticks. I will continue to test the different Hz and report back which setting worked best for my setup. Thanks again guys!!

closed 2025-03-28 by github-actions[bot]